RobTillaart/RS485 #10: Pin assignments

jackjean88, 2023-04-27:
Following a tip from a forum member, I downloaded your RS485-master library.
I am a real newbee when it comes to device communication.
I am working on a project with 1 Mega 2560 as master and 4 Nano V3 as slaves. I chose this technology because of the distance (~30m).
From the start I have some questions:
Assuming that the "sendPin" declaration concerns the pin connected to RE / DE of the MAX485 module, how can we define the pins for the RX and TX serial link (connected to DI and RO)?
For the master the send/receive switch pin is 2 and the RX and TX are 19/18.
For the slaves, depending on the units, the send/receive switch is various (4, 7 or 8), same for RX / TX which can be 5 / 6 or 6 / 7.
Thanks for your help.
J.

RobTillaart, 2023-04-27:
Thanks for the issue,
The rs485 modules all connect in same way like in this image. 

https://europe1.discourse-cdn.com/arduino/original/4X/7/d/c/7dcb2bc86398333312602a523206a18aca44fb3f.jpeg

To make it robust use a hardware Serial as those buffer incoming bytes in the background. 

Another point of attention is to disconnect the RS485 modules when uploading new code to a board.

RobTillaart, 2023-04-27:
For the mega2560, which has 4 hardware serial ports I would use Serial2, Serial3 or Serial4.

RobTillaart, 2023-04-27:
The pins of the hardwateSerials are hardcoded. You do not define them. You only pass the address of the Serial object to the rs485 constructor.

Have s look st the example master and slave.

jackjean88, 2023-04-27:
Thanks for your fast reaction .
For the Mega, I am wondering how to hardcode one Serial Port between 3 , since I already use the Nr 2 and 3 for another connections...
For the Nano(s) I defined myself the RX/TX pins, and you confirm that any SoftSerial configuration is running unless to declare it ?
I already run the Master example sketch, that returns a following of garbage characters (rectangle symbol)
The Nano as slave does not make flash the on board LED...

RobTillaart, 2023-04-27:
To learn to work with rs485, you should first do a mini project. One master one slave, keep it as simple as possible. If that works you can integrate it into a larger project. Trying to include it into a large project without understanding is possible but not advised. 

 Q: which Serial do you want to use for the Mega?

jackjean88, 2023-04-27:
I fully agree with you, that is why I pulled out from my big project one Mega and one Nano + 2 x Max485 modules. I simply tested your 2 first example sketches.
On the master, the Serial monitor displays  that means a problem.  If I replace the `cmd = random(3);` by `cmd = "abc";` the displayed status is "f", thats probably means OK.
I use the Serial1 on the Mega and a SoftwareSerial on the Nano.

jackjean88, 2023-04-27:
It is not mine, it is yours !
`void loop()
{
  if (millis() - lastCommand >= 2000)
  {
    lastCommand = millis();
    char cmd = random(3);     //  0,1,2
    rs485.print(cmd);
  }
  if (rs485.available() > 0)
  {
    int status = rs485.read();
    Serial.println(status);
  }
}
`
adding `byte incomingByte = Rs485.read();` and `Serial.println(incomingByte, HEX);` does not change, and does not display a line feed on the monitor

jackjean88, 2023-04-27:
Now it looks OK, displaying a random chain of A B or C. But it runs as standalone, because powering the slave or not does not change, and nothing occur on the Slave. It is like the master talks to himself...
I would highly appreciate getting a commented version of your both sketches to support my ignorance ...

jackjean88, 2023-04-27:
Could you please also tell me how can I dedicate a message to a given slave ? I do not find where this ID can be set...

RobTillaart, 2023-04-27:
Please note the library is still labeled **experimental**, it works quite well within my projects.
Is essence RS485 is transferring just bytes. Any protocol has to be developed on top of it.

> Could you please also tell me how can I dedicate a message to a given slave ? I do not find where this ID can be set...

You can set the ID in the constructor. 
```cpp
RS485(Stream * stream, uint8_t sendPin, uint8_t deviceID = 0);
```

The last parameter sets the device ID which is default 0,
This library provides an experimental **send()** and **receive()** with a message protocol that uses this ID.

You must:
- use ID = 0 for the master.
- use a different ID for every slave.

In a typical scenario the protocol starts with all slaves + master in receive mode.

The master sends a message with **send()** which includes the ID for the receiver.
After sending the whole message it goes into receive mode. 

Every slave will read every message send on the RS485 bus - using **receive()**
If looks at the right byte in the message stream for the device-ID. 
If that device ID matches the local device ID, it needs to process the message.
If that device ID does not match the local device ID it must ignore the message, 
Thereafter the slave waits for the byte that indicates the start of a new message again.

Assuming the slave must send a response to the master as part of processing the message:
The master is in receive mode and if a message comes in with the deviceID of the master in 
the right byte, the master can process the answer.

That is the idea how it works.

RobTillaart, 2023-04-27:
> I would highly appreciate getting a commented version of your both sketches to support my ignorance ...

I understand your need, however time is my scarcest resource. 
I need to rebuild a test setup and run the sketches again. 

-----

> Now it looks OK, displaying a random chain of A B or C. But it runs as standalone, because powering the slave or not does not change, and nothing occur on the Slave. It is like the master talks to himself...

mmm, then there is a problem in the demo sketches. I have a quick look at them.
(I have other tasks this evening so I cannot spend too much time on it now)

RobTillaart, 2023-04-27:
Try this for master

```cpp
//    FILE: RS485_simple_master.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: simple master
//     URL: https://github.com/RobTillaart/RS485

#include "Arduino.h"
#include "RS485.h"

const uint8_t sendPin  = 4;
const uint8_t deviceID = 0;

RS485 rs485(&Serial, sendPin);  //  uses default deviceID

uint32_t lastCommand = 0;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);

  rs485.setMicrosPerByte(115200);
}

void loop()
{
  if (millis() - lastCommand >= 2000)
  {
    lastCommand = millis();
    char cmd = '0' + random(3);     //  sends 0,1,2 as character  (not as value)
    rs485.print(cmd);
    Serial.print("SEND: ");
    Serial.println(cmd);
  }
  if (rs485.available() > 0)
  {
    int status = rs485.read();
    Serial.print("RECV: ");
    Serial.println(status);
  }
}

// -- END OF FILE --
```

---------------

and use this as slave

```cpp
//    FILE: RS485_simple_slave.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: simple listening slave
//     URL: https://github.com/RobTillaart/RS485

#include "Arduino.h"
#include "RS485.h"

const uint8_t sendPin  = 4;
const uint8_t deviceID = 1;

RS485 rs485(&Serial, sendPin, deviceID);

const uint8_t LED = 13;
uint8_t status = LOW;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);

  rs485.setMicrosPerByte(115200);

  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
}

void loop()
{
  if (rs485.available() > 0)
  {
    int c = rs485.read();
    Serial.print("RECV: ");
    Serial.println(c);

    if (c == '0') status = LOW;    
    if (c == '1') status = HIGH;
    if (c == '2')
    {
      Serial.print("SEND: ");
      Serial.println(status);
      rs485.print(status);
    }
    digitalWrite(LED, status);
  }
}

// -- END OF FILE --
```

It should show what is send and received on both sides.

jackjean88, 2023-04-27:
1 - I fully agree with your description of the RS485 based protocol.
2 - Please do not feel "forced" to spend your time with my subject if you consider being somewhereelse busy.
3 - I have updated both sketches with your changes.
     My remarks : 
it is strange not finding somewhere in the sent message the slave's ID (1 in this case).
Stil nothing arrives to the slave
The echo of what should be sent is strange : 3 strings are random displayed "1SEND: 1" or "2SEND: 2" or "3SEND: 3"
I do not explain why the random result is displayed first and followed by the char content...
in the line "char cmd = '0' + random(3);" the '0' never appears on the display. I can change it for anythingelse, no change.

RobTillaart, 2023-04-27:
I'll need tomake a test setup to see in detail what happens. If time permits tomorrow or coming weekend 
Starts to sound like a real bug so it is getting more priority.

RobTillaart, 2023-04-27:
The output you post is strange. I expected only SEND or RECV messages. Mmm curious...

RobTillaart, 2023-04-27:
Think I figured out where the echo cones from.
The sketch uses Serial as HW for the RS485 so all data written by the library pops up in your Serial monitor.

The next I dont understand is why it generates 123 instead of 012. Tomorrow.

RobTillaart, 2023-04-28:
Wrote a minimal RS485 sniffer, to see what goes over the RS485 bus.
Only connect the RX (receive) pin.
Will be part of the examples soon. 

```cpp
//    FILE: RS485_sniffer.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: simple sniffer
//     URL: https://github.com/RobTillaart/RS485

#include "Arduino.h"
#include "RS485.h"

const uint8_t sendPin  = 4;  // not used
const uint8_t deviceID = 250;  //  not used, maybe in future.

//  use a 2nd Serial port.
//  or only connect the RX of the MAX485.
RS485 rs485(&Serial, sendPin, deviceID);

const uint8_t LED = 13;
uint8_t status = LOW;

uint32_t count = 0;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println();
  Serial.println(__FILE__);

  rs485.setMicrosPerByte(115200);

  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
}

void loop()
{
  if (rs485.available() > 0)
  {
    //  toggle on receive
    digitalWrite(LED, !digitalRead(LED));

    if (count % 160 == 0)
    {
      Serial.println();
    }
    if (count % 16 == 0)
    {
      Serial.println();
      Serial.print(count);
      Serial.print('\t');
    }
    if (count % 4 == 0)
    {
      Serial.print(' ');
    }
    count++;

    int x = rs485.read();
    if (x < 0x10) Serial.print('0');
    Serial.print(x, HEX);
    Serial.print(" ");
  }
}

//  -- END OF FILE --
```

Going to write 2 additional samples that use the ID asap

jackjean88, 2023-04-28:
On that line : "RS485 rs485(**&Serial, sendPin**, deviceID);"

how can I select my **Serial1** port pins ? (or only RX pin)

My application(s) is (are) "hard wired" (printed boards), so I cannot change the connections.

jackjean88, 2023-04-28:
Sorry : the master sends thru Serial1 and the slaves receive on different pins configured by SoftWareSerial

RobTillaart, 2023-04-28:
Too busy, might take some time.

RobTillaart, 2023-04-28:
Created a develop branch in which the **send()** and **receive()** protocol is fixed. (still experimental).
Build a simple master slave configuration with it and that worked. 
master addressed 3 slaves and one was just ignored.
examples are included

Sniffer example is included

Will look at it later with a fresh pair of eyes before creating a PR.

RobTillaart, 2023-04-28:
> On that line : "RS485 rs485(**&Serial, sendPin**, deviceID);"
> 
> how can I select my **Serial1** port pins ? (or only RX pin)

For Serial1 use:
```cpp
RS485 rs485(**&Serial1, sendPin**, deviceID);
```

Hardware Serial ports have hard coded pins, you cannot change them.

> 
> My application(s) is (are) "hard wired" (printed boards), so I cannot change the connections.

Apparently you skipped a verification during design. 
I do not know your boards, but you cannot change the pins of hardware Serial's

You might give it a try with software serial but be sure you won't miss data you need a SW serial library 
that uses interrupts or so. I will see if I can do a quick test with the default SoftwareSerial.

RobTillaart, 2023-04-28:
> Sorry : the master sends thru Serial1 and the slaves receive on different pins configured by SoftWareSerial

OK,

RobTillaart, 2023-04-28:
Output from the master with SWS @9600 baud  (my testcode, will be in the develop branch soon)

```
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV: TEM: 22
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV: TEM: 22
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV: TEM: 15
```

Only 1 in 3 temperature requests are handled correctly, don't know the cause yet.
Humidity is not handled as that is on another slave (which is not connected)

Hedalen, 2023-04-28:
aybe, a stupid question,
but you talk about sendpin 4
but what to do with that pin, bcz i dont see it the schema?
https://europe1.discourse-cdn.com/arduino/original/4X/7/d/c/7dcb2bc86398333312602a523206a18aca44fb3f.jpeg

9600, 2023-04-28:
You called?

RobTillaart, 2023-04-28:
@9600 

Oops, sorry I wrote in a post about softwareserial at 9600 and wrote that as @ 9600 (without the space).
My error, apologies for the inconvenience.

Hedalen, 2023-04-29:
I dont understand, why it works with the uno and not with the mega
neighter as master and slave

RobTillaart, 2023-04-29:
> I dont understand, why it works

What do you exactly mean by **it** ?  Can you post your sketch?
(busy today with another library)

Hedalen, 2023-04-29:
i use this one as master
``//    FILE: RS485_simple_master.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: simple master
//     URL: https://github.com/RobTillaart/RS485

#include "Arduino.h"
#include "RS485.h"

const uint8_t sendPin  = 4;
const uint8_t deviceID = 0;

RS485 rs485(&Serial, sendPin);  //  uses default deviceID

uint32_t lastCommand = 0;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);

  rs485.setMicrosPerByte(115200);
}

void loop()
{
  if (millis() - lastCommand >= 2000)
  {
    lastCommand = millis();
    char cmd = '0' + random(3);     //  sends 0,1,2 as character  (not as value)
    rs485.print(cmd);
    Serial.print("SEND: ");
    Serial.println(cmd);
  }
  if (rs485.available() > 0)
  {
    int status = rs485.read();
    Serial.print("RECV: ");
    Serial.println(status);
  }
}

// -- END OF FILE --``

and this one as slave
``//    FILE: RS485_simple_slave.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: simple listening slave
//     URL: https://github.com/RobTillaart/RS485

#include "Arduino.h"
#include "RS485.h"

const uint8_t sendPin  = 4;
const uint8_t deviceID = 3;

RS485 rs485(&Serial, sendPin, deviceID);

const uint8_t LED = 13;
uint8_t status = LOW;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);

  rs485.setMicrosPerByte(115200);

  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
}

void loop()
{
  if (rs485.available() > 0)
  {
    int c = rs485.read();
    Serial.print("RECV: ");
    Serial.println(c);

    if (c == '0') status = LOW;    
    if (c == '1') status = HIGH;
    if (c == '2')
    {
      Serial.print("SEND: ");
      Serial.println(status);
      rs485.print(status);
    }
    digitalWrite(LED, status);
  }
}

// -- END OF FILE --

RobTillaart, 2023-04-29:
I will search for my Mega asap to reconstruct that setup.

Have you tried the Mega with Serial2 or 3 ?

Hedalen, 2023-04-29:
i used the serial0 but tried also the serial1, 2 and 3

RobTillaart, 2023-04-29:
Mmm, let me think.

Given that the Uno does work is an indication that the library works. We must find a way to be sure that the Serial ports of the Mega do work.

The library can be used to connect two Rs232 ports. You need to cross connect. TX to RX and GND to GND.

The sendpin does not have a job then. And it leaves the MAX485 modules out of the setup.

If you connect the Mega to the Uno

Hedalen, 2023-04-29:
for the crossed rs232 ports on the mega, can i use the rx0 and the tx0?

RobTillaart, 2023-04-29:
Yes,

RobTillaart, 2023-04-29:
Found my Mega and had it connected via Serial3 (and Serial2) to a NANO which had the SW serial sketch.
Baudrate 38400, 

See the same problem as with two SW serials. 
```
RS485_master_send_receive.ino
SEND: Get Temperature
RECV:	1	TEM: 23
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
SEND: Get Temperature
RECV:	1	TEM: 20
SEND: Get Humidity
SEND: Get Temperature
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV:	1	HUM: 55
SEND: Get Temperature
SEND: Get Temperature
SEND: Get Humidity
RECV:	1	HUM: 53
SEND: Get Temperature
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV:	1	TEM: 20
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
```

RobTillaart, 2023-04-29:
MEGA with Serial2 connected to NANO with Serial.
38400 baud

```
SEND: Get Temperature
RECV:	1	HUM: 57
SEND: Get Temperature
RECV:	1	TEM: 24
SEND: Get Temperature
SEND: Get Humidity
RECV:	1	HUM: 53
SEND: Get Temperature
SEND: Get Temperature
RECV:	1	TEM: 23
SEND: Get Humidity
SEND: Get Temperature
RECV:	1	TEM: 15
SEND: Get Temperature
SEND: Get Humidity
SEND: Get Temperature
RECV:	1	HUM: 52
SEND: Get Temperature
RECV:	1	TEM: 19
SEND: Get Temperature
SEND: Get Humidity
RECV:	1	HUM: 58
SEND: Get Temperature
SEND: Get Temperature
RECV:	1	TEM: 18
```

Still missing some requests, 
Think I have to check the parser when receiving data.

RobTillaart, 2023-04-29:
Yep, bug in the protocol.
MEGA Serial2 - Nano Serial works like a charm. Next test is SW Serial.
```
17:31:35.184 -> SEND: Get Temperature
17:31:35.184 -> RECV:	1	TEM: 19
17:31:38.171 -> SEND: Get Temperature
17:31:38.171 -> RECV:	1	TEM: 21
17:31:41.170 -> SEND: Get Temperature
17:31:41.170 -> SEND: Get Humidity
17:31:41.170 -> RECV:	1	TEM: 15
17:31:41.170 -> RECV:	1	HUM: 53
17:31:44.170 -> SEND: Get Temperature
17:31:44.170 -> RECV:	1	TEM: 17
17:31:47.169 -> SEND: Get Temperature
17:31:47.169 -> RECV:	1	TEM: 21
17:31:48.153 -> SEND: Get Humidity
17:31:48.153 -> RECV:	1	HUM: 59
```

RobTillaart, 2023-04-29:
Mega Serial2 - Nano SWSerial 38400
substantial better than before, only misses a call when 2 requests are made very short after each other. 
But this is expected.
```
17:34:58.943 -> SEND: Get Temperature
17:34:58.943 -> RECV:	1	TEM: 23
17:35:01.943 -> SEND: Get Temperature
17:35:01.943 -> RECV:	1	TEM: 15
17:35:03.952 -> SEND: Get Humidity
17:35:03.952 -> RECV:	1	HUM: 52
17:35:04.936 -> SEND: Get Temperature
17:35:04.936 -> RECV:	1	TEM: 19
17:35:07.935 -> SEND: Get Temperature
17:35:07.935 -> RECV:	1	TEM: 23
17:35:10.935 -> SEND: Get Temperature
17:35:10.935 -> SEND: Get Humidity
17:35:10.935 -> RECV:	1	TEM: 18
17:35:13.929 -> SEND: Get Temperature
17:35:13.929 -> RECV:	1	TEM: 24
17:35:16.928 -> SEND: Get Temperature
17:35:16.928 -> RECV:	1	TEM: 15
17:35:17.913 -> SEND: Get Humidity
17:35:17.959 -> RECV:	1	HUM: 55
```

Going to clean up code / examples and will push latest version to the develop branch.

Hedalen, 2023-04-30:
> Yes,

It seems, the serial ports at the mega works okay, bcz the tx and rx leds lights up

RobTillaart, 2023-05-01:
@Hedalen 
Any progress made?

Hedalen, 2023-05-01:
I changed all to 9600, now the mega works okay
But still i dont the send and receive, bcz they are not the same

RobTillaart, 2023-05-01:
> But still i dont the send and receive, bcz they are not the same

What you mean by "they are not the same"? 
Who/what is they?
Can you elaborate on the difference ?

I've done several test (see above) and especially the SWSerial were missing requests. 
The HWSerial version are pretty stable imho.

Hedalen, 2023-05-02:
mean, the numbers (1, 2, 3) the master send, i dont see that number in the
serial monitor from the slaves

RobTillaart, 2023-05-02:
Do you send the 1,2,3 as integers or as character?

RobTillaart, 2023-05-02:
I cannot explain,
What board you have on the receiving side?
Are there leds flashing on the receiving side.
Have you tried communicating just over RS232 (without the RS485 devices)?

Hedalen, 2023-05-02:
i have on the receiving side one uno and two megas
one mega have flashing leds and one dont have flashing leds
then, i have take an other mega for the not flashing mega, but no changes
over rs232, both flashing
i even try to swapp the rs485, but no results

RobTillaart, 2023-05-02:
So if I understand correctly the communication works when not using the RS485 modules?
(sorry for the specific questions, but I have to make a mental model of what happens (or not).

RobTillaart, 2023-05-02:
That implies the library works over RS232  (which I have confirmed before)

==> the problem has to be in the 485 modules?

Q1: Do you have a oscilloscope to see signals on the A and B lines of the RS485 module?
Q2: which RS485 modules do you have?

Hedalen, 2023-05-02:
A1, i dont have a scope, sorry
A2 i have the 
https://www.wish.com/product/5e0615b87d567e0b3c0489a2 modules

RobTillaart, 2023-05-02:
Scope is not the most important tool. (1 = multimeter, 2 = logic analyzer 3 = scope imho)

OK that are standard modules.

Q: Did you connect (both modules)
- the **sendPin** (defined in constructor to BOTH the RE and DE pins of the module?
- the TX of the board to the DI
- the RX of the board to the RO
- VCC + GND correctly (if you have swapped them the chip may burn)

The modules need to be connected 
- A to A 
- B to B

Hedalen, 2023-05-02:
all 4 modules connected on the same way
2 unos and 2 megas
https://europe1.discourse-cdn.com/arduino/original/4X/7/d/c/7dcb2bc86398333312602a523206a18aca44fb3f.jpeg

RobTillaart, 2023-05-02:
Need some time to think about what could cause this. 
(this might take a few days, as other tasks are pending)

Hedalen, 2023-05-02:
BTW, the master send very good the random 0, 1, 2 characters
i saw that in the serial monitor

RobTillaart, 2023-05-02:
OK, but we do not know if they get correctly into the RS485 module.
Do you have an LED and a 1K resistor?
If you put them in series you might see the pulses on the AB lines, especially at lower baudrates.

RobTillaart, 2023-05-02:
Idea
```cpp
  inline void setTXmode() { digitalWrite(_sendPin, HIGH); };
  inline void setRXmode() { digitalWrite(_sendPin, LOW); };
```
If we put the master which sends the digits in a constant transmit mode, and the receiver in a constant receive mode?

- constant sending means connect the RE and DE pins to 5V
- constant receiving means connect the RE and DE pins to GND

This way one can check the working of the **sendPin**.

Hedalen, 2023-05-02:
> OK, but we do not know if they get correctly into the RS485 module. Do you have an LED and a 1K resistor? If you put them in series you might see the pulses on the AB lines, especially at lower baudrates.

i have do this, but put the led with the resistor on the DI and the RO from the modules, then they blinck only at the modules, mounted to the unos, not to the megas
i think thats strange, bcz one mega shows me that the RX blinks
they blink not when i put them on the A and the B from the modules

RobTillaart, 2023-05-02:
Could it be a broken wire?

Hedalen, 2023-05-02:
> Idea
> 
> ```c++
>   inline void setTXmode() { digitalWrite(_sendPin, HIGH); };
>   inline void setRXmode() { digitalWrite(_sendPin, LOW); };
> ```
> 
> If we put the master which sends the digits in a constant transmit mode, and the receiver in a constant receive mode?
> 
> * constant sending means connect the RE and DE pins to 5V
> * constant receiving means connect the RE and DE pins to GND
> 
> This way one can check the working of the **sendPin**.

and where do i put that in my sketch?

RobTillaart, 2023-05-02:
You cant put that in the sketch. 
Just use wires to directly.

[20 more replies]

closed 2023-05-11 by RobTillaart