RobTillaart/RS485 #14: send, receive, ACK, NAK

rozrabiak, 2023-08-04:
Hi! I have additional question.

In file RS485.cpp i found:
```
//  optional
//    ACK      0x06   ACKnowledge
//    NAK      0x15   Not Acknowledge
```
Is there any way to use this in this scenario:
sending data from slave -> master receive and send to slave ACK when received -> slave send function return TRUE but when master not receive (no ACK from master or master return NAK) then on slave send function return FALSE.

And that same when master send data to slave then ACK/NAK and send function return TRUE or FALSE.

Thank's.

RobTillaart, 2023-08-04:
This is how it typical works, but you are free to define your own protocol

The master is in control.
So the slave could send an ACK or NAK (if it cannot handle the request but is able to answer) to inform the master.

The master has no obligation to inform the slave.
In fact the slave may assume that the data he sent did arrive at the master. 
Otherwise the master will request it again.

----

If you want to inform the slave with a ACK that the data has been received, should the master not know that his ACK has been received by another ACK from slave to master, which needs to be confirmed by another ACK from master to slave .... ad infinitum.

RobTillaart, 2023-08-21:
Assuming this problem is solved, I close the issue.

closed 2023-08-21 by RobTillaart