RobTillaart/RS485 #15: Why no _stream->flush(); used?

wltue, 2023-12-05:
I had issues with the slave, It response so fast the Master didn't put the MAX485 in receive mode yet, missing the first byte of the slave sending.
I changed:
"delayMicroseconds((len + 8 + 2) * _microsPerByte); //  + 2 to be sure..."
in
"void RS485::send(uint8_t receiverID, uint8_t msg[], uint8_t len) {}"

to

"_stream->flush();"

Works perfectly on 2 baud rates, master and slave. Verified on a DPO2024B. Why are timers used?
The master is a Arduino Nano ESP32 with a MAX3471CUA+T.
The slave is a Arduino Zero with a MAX3471CUA+T.
Finally a nice layer library for RS-485. Thanks for the lib.

with "delayMicroseconds((len + 8 + 2) * _microsPerByte); //  + 2 to be sure..." (the blue is the _sendPin)
[image]

with "f_stream->flush();"
[image]

RobTillaart, 2023-12-05:
Thanks for this issue.

I used delayMicroseconds() to be sure to wait long enough - as throughput can be calculated.
Apparently the latter part (about calculation) is an assumption on my side as I assumes a certain throughput.

I will check the proposed change **flush()** as it looks like a more robust way to handle it.

>  Finally a nice layer library for RS-485. Thanks for the lib.

Thankyou and you're welcome!

RobTillaart, 2023-12-05:
had a quick look and it looks good. There are 2 more places to patch the code.

It could imply that 
```cpp
  //  baudRate must match Serial baudRate
  //  micros are needed before RS485 stream may be reset
  //  to (default) receiving mode.
  void     setMicrosPerByte(uint32_t baudRate);
  uint32_t getMicrosPerByte();
```
is not needed anymore thereby simplifying the code.

wltue, 2023-12-05:
Will test Asap.

RobTillaart, 2023-12-05:
Created a develop branch to prepare version 0.3.0 with your proposal.
Build-CI is running.

RobTillaart, 2023-12-05:
PR created, build runs.

wltue, 2023-12-06:
Works perfect!
I saw you removed the timed versions, but it could be "flush" doesn't work on other devices as I tested.
I tested it on 38400 and 57600 baud.
Thank you!

overview
[image]

zoomed in switching from tx to rx
[image]

zoomed in last byte from slave sending, hex
[image]

zoomed in last byte from slave sending ascii
[image]

RobTillaart, 2023-12-06:
> Works perfect!

good to hear, 

>  but it could be "flush" doesn't work on other devices 

The build-CI test it on 5 platforms so I expect it will work.   (see .arduino-ci.yml)
    - uno
    # - due
    # - zero
    # - leonardo
    - m4
    - esp32
    - esp8266
    # - mega2560
    - rpipico
    
If flush is not supported, users should report this, and I will search for a new solution (conditional code)

RobTillaart, 2023-12-06:
NOte: I cannot view the images in detail => errror 

## This private-user-images.githubusercontent.com page can’t be found

with a large URL,

RobTillaart, 2023-12-06:
I will merge the PR asap.

commit 9a624d22 (2023-12-06): Fix #15 - use stream.flush() (#17)

- Fix #15 - use flush() instead of calculation.
- remove **setMicrosPerByte()** and **getMicrosPerByte()**
  - not needed any more.
- optimize **write(array, length)** when yield flag is set
closed 2023-12-06 by RobTillaart