RobTillaart/RS485 #18: Request for an example to how to use ACK and NACK

wltue, 2023-12-06:
It would be awesome if you could add an example for NACK and ACK. How this should be implemented as for seen in the RT RS-485 protocol :-)

RobTillaart, 2023-12-06:
ACK and NACK are (byte size) signals between two communicating nodes.
The ACK just confirms that a packet (1 or more bytes) is received correctly.

What correctly means is protocol dependent. It might mean just that a byte has been received or more complex that a CRC or other checksum, or the length of the packet etc has been confirmed. 

Example if the master sends an array of  8 bytes it might send
[ 8 A B C D E F G H ]   8 being the length

If the slave receives 
[ 8 A B C D E F G H ]
it will respond with 
[ACK]

If the slave receives 
[ 8 A B C D A B C D ]
it will respond (incorrectly) with 
[ACK]

If the slave receives 
[ 8 A B C D  ]
it will respond (correctly) with 
[NACK]

wltue, 2023-12-06:
Thanks very clear. In this project the messages need to be fail safe. Working on it. Thank you again for your library and help!

RobTillaart, 2023-12-06:
Find attached a non-tested pair of sketches to allow one Arduino control a second one (partially).
I revived some old code but have no hardware free to test it properly.
It might be added in time as a pair of extra examples.

[examples.zip](https://github.com/RobTillaart/RS485/files/13590373/examples.zip)

closed 2023-12-18 by RobTillaart