RobTillaart/RS485 #21: Sometimes RS485_master_send_receive.ino sends while recieving

wltue, 2023-12-13:
Haai
In RS485_master_send_receive.ino

The sending action will have to wait for a recieve. using:
if (millis() - lastT >= 3000)
and
if (millis() - lastH >= 7000)
will send at some point the same time while recieving, showing really funny signals on the Oscilloscope :-)

I have a quick and dirty solution:

In setup
bool_Recieved = true;

In loop:
if ((millis() - lastT >= 3000) && (bool_Recieved == true))
{
  ...
  bool_Recieved = false;
}

if ((millis() - lastH >= 7000) && (bool_Recieved == true))
{
  ...
  bool_Recieved = false;
}

and 

 if (rs485.receive(ID, arr, len))
{
  ...
 bool_Recieved = true;
}

I think it's needs more like a state machine solution. The master needs always wait for response after sending. But still if the CRC is incorrect the slave will not respond and I guess the program will lockup? So also a timeout option. 
Thank you for reading.

RobTillaart, 2023-12-13:
Thanks for your remarks, they are as always welcome.
Do you use the latest 3.0 version already?

In the end it all depends on the protocol how things should behave. 
The examples are meant to get people started, not as an 100% engineered solution.

RobTillaart, 2023-12-13:
Q: is this related to the issue  #18 ?

wltue, 2023-12-13:
> Q: is this related to the issue #18 ?

Not related.

wltue, 2023-12-13:
> Thanks for your remarks, they are as always welcome. Do you use the latest 3.0 version already?
> 
> In the end it all depends on the protocol how things should behave. The examples are meant to get people started, not as an 100% engineered solution.

0.3.0 Yes. I just report. You can do everything what you want with it :-). For me its stable a.t.m. Thanks.

closed 2023-12-13 by RobTillaart