RobTillaart/RS485 #23: Making RPi device as Master Controller

VigneshVicky97, 2023-12-22:
Hi @RobTillaart ,
   Thanks for giving me such great library on RS485 communication. I have been working with your RS485 library for more than 2 months. I successfully connected 30 RS485 devices on a Single bus and communicating with each other with your library. I used ESP32 for all the devices. Now I want to convert the Master device into Raspberry Pi. I tried to convert your code into Python. In the latest version (0.4.0) of your library, you have mentioned that you used the RPi Pico board in RS485 Slave Controller Example program [.arduino-ci.yml](https://github.com/RobTillaart/RS485/blob/master/examples/RS485_slave_controller/.arduino-ci.yml) file. Can you please explain the process of implementing your library on RPi Pico board and RPi 4B board ?

   On seeing the other issues in your library, I think you are trying to implement the ACK and NACK in the communication protocol for more robust communication and finding the surety of message receiving from the Slave devices. Do you have any plans of implementing those names in the message [format](https://github.com/RobTillaart/RS485/blob/7b5b3c267ad0a03fd1323f3dd5e98aa63f650369/RS485.cpp#L133) on your library?

RobTillaart, 2023-12-22:
Thanks for the compliment, you're welcome.

----
wrt your questions

## RPi Pico
This board is programmable with the Arduino IDE in C++. Check the board manager.

## RPI4B
The library is never tested on RPI as far as I know. it should be possible to port the code to the C++ environment of RPI.

## Python
Sorry, I do not have time to port my libraries to RPI / python. Feel free to port it and reference this library (see license).

## ACK / NACK
The examples are just meant for people to have a starting point. 
There was a request for an ACK/NACK example so I made a very simple one. 
Imho ACK/NACK has very limited use.

The message format in the library is a more worked out example.
I have no intention to extend the protocol with ACK NACK.
Users are free to implement their own (derived) class and override the protocol.

Hope this answers your questions.

VigneshVicky97, 2023-12-22:
Thanks for your Quick and Clear explanation. I will try to implement the library in RPi 4 and let you know, if I succeed. Keep up your good work.

closed 2023-12-22 by VigneshVicky97

RobTillaart, 2023-12-22:
If you make a Python repository, I will link to that of course!