RobTillaart/RS485 #24: Sending uint64_t size Message

VigneshVicky97, 2024-01-03:
Hi @RobTillaart ,

I want to send the Unique MAC ID (uint64_t) of the ESP32 devices to the Master Controller using your library through RS485. When I try to send data more than 8 digits, it won't reach the receiver. How can I achieve this? Do you have any solution for this?

RobTillaart, 2024-01-03:
Thanks for your question.
Not knowing your code, I assume you have to flush the send buffer of the slave before you set the slave to listening mode.

RobTillaart, 2024-01-09:
@VigneshVicky97 
Have you tried flush?

VigneshVicky97, 2024-01-11:
Thanks for your suggestion. Now I am able to get the full message without any loss. But I have some other doubts on your code. I am using your code for my office project. But I am not sure, I can share that code in this page. Is there any other way, I can solve my doubts without posting online?

closed 2024-01-11 by VigneshVicky97