RobTillaart/RS485 #25: Raspberry Pi Pico support

DoomHammer, 2024-01-04:
Hi!

Thanks for creating this neat library! I started using it with my Pi Pico based project and I was wondering if you actually support this hardware and test your releases on it?

RobTillaart, 2024-01-04:
Thanks for your question.
It is impossible for me to test all my 200++ libraries on all platforms. 
Only if someone would pay me for it I could stop working and do full time library development and testing. 

During development I test with Arduino UNO or MEGA (e.g. when second HW Serial is needed) and ESP32.
These platforms are also configured in the build-CI script of most libraries so I will at least catch compile errors.
Recently I changed in several libraries that use SPI to become easier to run on RP2040 as it is not compatible there.

Also for the I2C libraries I made changes as settings the SCL/SDA pins is different on different platforms,
Although I initially supported this in the library I removed it as it is not a task / responsibility of the library.
This gave a cleaner dependency injection in those classes. 

I recall testing this library with MEGA and a few UNO's with 10+ meters of cable. There was a problem in the protocol that took some analysis to understand what happened. In the end the cause was that the answer of one slave was interpreted by another slave as a request for info, causing two or three slave going in TX mode.

Wrt your question, I do not recall testing it with an RP2040. 
My aim is to support the RP2040 and others by keeping board specific code (e.g. low level registers) from the library, or make it explicit where such code resides. (e.g. FastShiftOut library for AVR).

Any more questions?

DoomHammer, 2024-01-05:
I'm sorry if my questioned seemed rude.

By no means I was suggesting you should test your library on all the possible hardware out there.

I was merely curious what platforms you do test on so if anything goes wrong we could "compare notes".

If you don't test on RP2040 I assume this platform is unsupported, so any problems I might have are my own and if I solve them, the best I can do is to open a PR. If this platform was "supported", so I knew you're actively using it, I'd share my woes earlier on hoping to get some insight from you (like "your pinout is wrong" or "you need to call `foo` first").

Thanks for your answer and I'm going back to figuring out what have I done wrong :)

RobTillaart, 2024-01-05:
>  I'm sorry if my questioned seemed rude.

No need to apologize, for me it did not feel rude at all, 

> If you don't test on RP2040 I assume this platform is unsupported, so any problems I might have are my own and if I solve them, the best I can do is to open a PR.

All libraries (at least they should) compile for RP2040 as it is an important platform. The fact that I do not run all examples and all tests on it does not mean it is unsupported. If a platform gives problems with any of my libraries I am willing to have a look at it and try to fix / patch / whatever, to make it working. So if you have problems with my libraries just open an issue and I see what I can do. If nothing I will tell you too.

Keeping code portable is still one of the most important goals. This has as side effect that the code is not always the fastest or smallest possible, but it gives far less maintenance problems. Think when you have 5 or more lookalike libraries, one per platform, maybe with specific code. And a new platform pops up, Which of the lookalikes should be used? Or do we get lookalike nr X?

So just open an issue if you have a problem and I will try to think and analyze with you.

DoomHammer, 2024-01-05:
After consulting with a friend it turned out that I had the wrong connection between the two devices and that's what was causing the problems.

The board I used (from Waveshare) has a RJ11 connector, a goldpin one (DuPont), and a screw terminal.

I wanted to use the RJ11 in the final piece so I simply assumed any RJ11 cable would be fine. It turns out the cable must be either crossed or uncrossed in an opposite way to what the designers designed. So the A and B lines were probably not where they should be.

When I connected the goldpin terminals directly both the low level (`read`) and high level (`receive`) interfaces seem to be working.

Thanks again for creating this awesome piece of software!

RobTillaart, 2024-01-05:
Good to hear it is solved and apparently nothing broke due to the wrong connections.
If there are no further questions, you may close the issue.

closed 2024-01-05 by DoomHammer