RobTillaart/RS485 #28: Need help

MatiM72737, 2024-02-10:
Is able to make it like master asking each of slaves about information, but on init all slaves automatically gets id without need to setup it manually

RobTillaart, 2024-02-10:
Thanks for your question.
As every slave id must be unique, you need to implement a dynamic id protocol. This is not supported in this library.

Designing and implementing such protocol is not trivial and hard to get foolproof. I would go for a variation of DHCP or Appletalk, maybe ARP. These protocols implement what you asking for.

Looking at DHCP, it is a server (master) that hands out addresses (id) on request. 
So at startup a rs485 slave must request its id from the master.

RobTillaart, 2024-02-10:
The master holds a list of free id s and after assigning one it is removed from the list.

In Dhcp the id is only valid for a time T and the 'lease' of the id need to be refreshed before time T has passed. alternative is a master that sends keep alive messages to every slave.

Hopes this helps to get started. I advice to study the working of the protocols mentioned.

RobTillaart, 2024-02-11:
As I will not implement this dynamic behavior in the library, I close this issue

closed 2024-02-11 by RobTillaart