RobTillaart/RS485 #29: Can't update in platformio (Arduino nano)

rozrabiak, 2024-03-03:
Hello Rob,
I can't update to latest version.
Info below:

[image]

RobTillaart, 2024-03-03:
Have to look into that

RobTillaart, 2024-03-03:
version 0.5.0 is published
```
pio package publish
Preparing a package...
Type:     library
Owner:    robtillaart
Name:     RS485
Version:  0.5.0
Size:     13.88KB
UserSideException: The package `robtillaart/RS485@0.5.0` is already published in the registry
```

[image]

So it is published on platformio, even 4 versions more.

I do not know why you can't see the latest, 
maybe you need to update your platformio environment firrst?

(you can always fetch direct from GitHub...

RobTillaart, 2024-03-10:
@rozrabiak 
Did you find the cause yet ?  (for future reference)
As this is not a library issue you may close the issue

RobTillaart, 2024-03-11:
As this is not a library issue I close the issue, feel free to reopen if you have more information.

closed 2024-03-11 by RobTillaart