RobTillaart/RS485 #30: RS485.write doesnt work on STM32, _stream.flush() not supported on STM32

r0n0x, 2024-04-03:
in between the calling of setTXmode and setRXmode to enable the driver for transmission, flush() is called to ensure that setRXmode is not called until the buffer has cleared and the transmission is over. the problem though is that in the stm32 framework, serial flush, or at least stream flush as is used here, does not work, the function does not get blocked and setRXmode is called immediately after the serial write begins 

[image]

this is the result, the DE pin is disabled before the first bit has even been transmitted. blue channel is the DE line, yellow is the microcontrollers TX->D

for this example an esp32 master was used and it had no problem doing the same thing properly. 

For STM32 some other block or delay would need to be implemented, because the flush() function is apparently empty and the name is just included to prevent compiler errors. 

Specific board this was used on was a STM32F401CEU6 blackpill.

RobTillaart, 2024-04-03:
Thanks for the issue,
Minimal experience with STM32, so needs investigation. Might take some days.

This seems to be the core problem / solution

> For STM32 some other block or delay would need to be implemented, because the flush() function is apparently empty and the name is just included to prevent compiler errors.

RobTillaart, 2024-04-03:
Thanks for the additional info:

HardwareSerial.cpp does implement **flush()** and it looks correct. See below
However I need to check the **_written** flag, if it is set correctly as that could explain a too fast return.

```cpp
void HardwareSerial::flush()
{
  // If we have never written a byte, no need to flush. This special
  // case is needed since there is no way to force the TXC (transmit
  // complete) bit to 1 during initialization
  if (!_written) {
    return;
  }

  while ((_serial.tx_head != _serial.tx_tail)) {
    // nop, the interrupt handler will free up space for us
  }
  // If we get here, nothing is queued anymore (DRIE is disabled) and
  // the hardware finished tranmission (TXC is set).
}
```

RobTillaart, 2024-04-03:
searched for _written in HardwareSerial.h/cpp  [1.9.0]

from HardwareSerial.h   (~100)
```cpp
    // Has any byte been written to the UART since begin()
    bool _written;
```

from HardwareSerial.cpp (~460)
```cpp
size_t HardwareSerial::write(uint8_t c)
{
  _written = true;
  if (isHalfDuplex()) {
```

So the flag is set to true, however it looks like it is never initialized.

RobTillaart, 2024-04-03:
Then I checked if I had the latest greatest STM32 environment ==>

[image]

So update needed.

RobTillaart, 2024-04-03:
OK, 2.7.1 looks different but the _written variable is still not initialized in the class.

HardwareSerial.cpp (~490)
```cpp
void HardwareSerial::flush(uint32_t timeout)
{
  // If we have never written a byte, no need to flush. This special
  // case is needed since there is no way to force the TXC (transmit
  // complete) bit to 1 during initialization
  if (_written) {
    uint32_t tickstart = HAL_GetTick();
    while ((_serial.tx_head != _serial.tx_tail)) {
      // the interrupt handler will free up space for us
      // Only manage timeout if any
      if ((timeout != 0) && ((HAL_GetTick() - tickstart) >= timeout)) {
        // clear any transmit data
        _serial.tx_head = _serial.tx_tail;
        break;
      }
    }
    // If we get here, nothing is queued anymore (DRIE is disabled) and
    // the hardware finished transmission (TXC is set).
  }
}
```

Two conditions that fall through **flush()**
- if **_written == false** it will return immediately,
- if there is a timeout  (needs investigation)

RobTillaart, 2024-04-03:
timeout is default 0 so that path is not used.

An interesting function could be 
```cpp
int HardwareSerial::availableForWrite(void)
{
  tx_buffer_index_t head = _serial.tx_head;
  tx_buffer_index_t tail = _serial.tx_tail;

  if (head >= tail) {
    return SERIAL_TX_BUFFER_SIZE - 1 - head + tail;
  }
  return tail - head - 1;
}
```

**flush()** could be mimicked with this function by following while loop.
(wait until whole buffer is ready for writing)
```cpp
while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1));  //  do nothing...
```

Can you change the RS485.cpp line 56 file and give it a try?

```cpp 
void RS485::flush()
{

  while (_stream->availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1)); 
}
```

r0n0x, 2024-04-04:
`while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1));  //  do nothing...` Works perfectly, however, i am not skilled enough to directly modify the library, so what i did instead was pull the rs485.send() function into my main. i had already done that actually, and it was working, but using a fixed delay() in the mean time. i swapped delay with the whole serial availableforwrite, and bam, works flawlessly, not a microsecond of lag there either or data clipping.

Using that blocking code, your library now works perfectly off stm32 hardware serial :)

r0n0x, 2024-04-04:
[image]
the time difference gap between the end of the master transmission and the start of the slave transmission is pretty sheer, you are right that delays are needed. the upside at least of now having re-written the send command in main is im free to also change that while into a timer driven if or case sequence, because now that i think about it, ill definitely be doing some stuff that requires realtime sensor polling.

RobTillaart, 2024-04-04:
Thanks for this feedback,  not read them all in detail yet.

it looks like we are going in the right direction.. 
The fixed delay() after the while loop must be made baud rate dependent. 

At 9600 baud it takes a ms to send a byte, so to send n bytes we need something like
Delay( 20000/baud + 1)

Can you try to change the while loop by removing the -1 from the test.? If that works it would reduce the needed delay with one (last) character to be sent. Note: if it blocks forever it is needed :)

r0n0x, 2024-04-04:
Just checked, removing the "-1" results in for some reason, only a single transmission to be successfully sent back, afterwards it seems to lock up

RobTillaart, 2024-04-04:
> Just checked, removing the "-1" results in only a single transmission to be successfully sent back, afterwards it seems to lock up

That means it is needed. OK we checked that corner case. 

What happens in the CPU is that the last byte is removed from the TX buffer and put in the transmit register. Then that last byte is clocked out including defined start (1) and stop (0..2) bits, followed by at least one bit pause (to separate the bytes).
That means a byte = 8 bits needs 10-12 pulses to transmit. So at 9600 baud one byte takes 11 / 9600 is about 1.15 millis.
Given that there might be variations in clocks one could use 1.2 millis.

The delay after the loop must be long enough to send that last byte. ==> **delayMicroseconds(11000000 / baud);**
That should be about the most optimized delay time.

wltue, 2024-04-04:
Using MAX3471CUA+T in my design, i saw some wierd delay times in switching on the scope. It needed just a tad tick more. Could be me and I had an other timer option available, so i played on safe for the delay and not the exact calculated time. Maybe the pullup, down terminator resistors in combination with cable length on the bus could be a thing in switchover time (shouldn't i know). Thanks all.

wltue, 2024-04-04:
I dont use your library atm, due to fact it has blocking code. I use taskscheduler library  for tasks and a seperate task switches over the rs485 tranciever, So putting the code will be an void, sorry. I was hoping you library had an update sooner or later. I was posting to help out.

RobTillaart, 2024-04-04:
@wltue 
> I dont use your library atm, due to fact it has blocking code. I use taskscheduler library for tasks and a seperate task switches over the rs485 tranciever, So putting the code will be an void, sorry. I was hoping you library had an update sooner or later. I was posting to help out.

Constructive help is always welcome, so thank you!

A non blocking version is not in my planning (maybe maintaining to much libraries ;) 
Are you familiar with - https://www.gammon.com.au/forum/?id=11428
It has some non-blocking code / comments.

wltue, 2024-04-04:
Don't worry about the update. Your library was excellent startup for the project.  I'ts very clear and understandable code. I always worked with CAN transcievers but had to build this project with RS485 and I am impressed how hard it is to adapt half duplex v.s. CAN bus.

RobTillaart, 2024-04-04:
@r0n0x 

Created a develop branch (0.5.1) with the following implementation for flush().

Can you test it with your STM32 ==> replace in the rs485.cpp file

If this works the code will be promoted to a PR.

```cpp
void RS485::flush()
{
  //  _stream->flush();
  while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1));
  delay(1);  //  for the last byte.
}
```

RobTillaart, 2024-04-04:
too bad, SERIAL_TX_BUFFER_SIZE is not standard over platforms

```
/home/runner/Arduino/libraries/RS485/RS485.cpp: In member function ‘virtual void RS485::flush()’:
    /home/runner/Arduino/libraries/RS485/RS485.cpp:59:41: error: ‘SERIAL_TX_BUFFER_SIZE’ was not declared in this scope
       59 |   while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1));
```

I have to find some #ifdef to detect the STM32 architecture.

```cpp
void RS485::flush()
{
  _stream->flush();
#ifdef STM32_SOMETHING
  while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1))
#endif
  delay(1);  //  for the last byte.
}
```

RobTillaart, 2024-04-04:
Found these #defines 
-DSTM32F4xx
-DARDUINO_BLACK_F407VE 
-DARDUINO_ARCH_STM32
-DSTM32F407xx

The 3rd seems the best an option as it is not board specific.

RobTillaart, 2024-04-04:
Updated the develop branch for test.

RobTillaart, 2024-04-04:
@r0n0x 

Can you change the flush() call in rs485.cpp with this one and verify it works?
```cpp
void RS485::flush()
{
#ifdef ARDUINO_ARCH_STM32
  while (Serial.availableForWrite() < ( SERIAL_TX_BUFFER_SIZE - 1));
#else
  _stream->flush();
#endif
  delayMicroseconds(_microsPerByte);  //  for the last byte.
}
```

RobTillaart, 2024-04-04:
FYI
added two functions:
- void setMicrosPerByte(uint16_t mpb) { _microsPerByte = mpb; };
- uint16_t getMicrosPerByte() { return _microsPerByte; };
- _microsPerByte = 1100 default.

RobTillaart, 2024-04-11:
@r0n0x 
Any progress to report?

RobTillaart, 2024-04-16:
@r0n0x 
As I got no answer I assume the problem is solved, so I merged the code into master and released a 0.5.1 version.
If there are still problems feel free to reopen this issue.

commit 49abd7df (2024-04-04): fix #30, flush() for STM32